marin-m/vmlinux-to-elf #5: Option for increasing rom size

DrRamm, 2020-01-05:
By default IDA suggest ROM size as Loading size, but it's not good. Better increase ROM size for handling virtual values, which in case ROM == Loading show as red.  Increasing ROM size will help to get rid of that.

p1-mmr, 2020-01-05:
Hello,

`vmlinux-to-elf` does not create a "ROM section", but rather a `.bss` section of fixed size in the generated ELF file. See: https://github.com/marin-m/vmlinux-to-elf/blob/d7f926d/vmlinux_to_elf/elf_symbolizer.py#L84

By default, this section starts right after a `.kernel` section (which contains the code and initialized data of the kernel, from the input file), and is of a size of `0x100000` (1 MB).

Can you please specify more the case that you encountered:

* Did you encounter references to data whose the addressing was located far outside of the `.kernel` and `.bss` sections, or
* Did you encounter references to data that was located somewhat after the end of the automatically created `.bss` section?

Using a command such as `readelf -aW` or `objdump -x` over the generated ELF file may be useful to you to analyze the layout of the generated file.

Thank you,

marin-m, 2020-01-05:
Hello,

`vmlinux-to-elf` does not create a "ROM section", but rather a `.bss` section of fixed size in the generated ELF file. See: https://github.com/marin-m/vmlinux-to-elf/blob/d7f926d/vmlinux_to_elf/elf_symbolizer.py#L84

By default, this section starts right after a `.kernel` section (which contains the code and initialized data of the kernel, from the input file), and is of a size of `0x100000` (1 MB).

Can you please specify more the case that you encountered:

* Did you encounter references to data whose the addressing was located far outside of the `.kernel` and `.bss` sections, or
* Did you encounter references to data that was located somewhat after the end of the automatically created `.bss` section?

Using a command such as `readelf -aW` or `objdump -x` over the generated ELF file may be useful to you to analyze the layout of the generated file.

Thank you,