marin-m/vmlinux-to-elf #16: Lots of invalid instructions when analyzed by ghidra and radrare2

DrmnSamoLiu, 2020-08-05:
First of all great project 👍 I can imagine this project will help tons of firmware researchers out there.

I've met a problem though. 
I'm currently doing research on a network camera firmware. Although binwalk didn't really identified vmlinux.img in the firmware analysis, I managed to found the portion of raw binary that is supposed to be the kernel image for the camera.
The data portion can be successfully analyzed and convert into ELF file by your script. However when I tired to use Ghidra to analyze it, it produces a lot of "invalid instruction" error. (Same in radare2)

The camera is running on a MIPS processor and your script have no problem identifying it, so I'm not sure what the problem might be.
Other binaries from the same firmware file can be analyzed without problem when setting language as `MIPS:LE:64:64-32addr:o32` in Ghidra.

The data portion I mentioned can be downloaded here : [https://drive.google.com/file/d/15gWN5dsWeiSefHpzh9VzPfwiUEpg_GKL/view?usp=sharing](url)

marin-m, 2020-08-05:
Hello,

Thanks for your feedback. I have processed your sample kernel with the tool and the ELF file seems to be reconstructed fine. The offsets are calculated correctly and the functions decompile with Ghidra or disassemble from beginning of their prologues with IDA..

Maybe that the Ghidra warnings that your see are due to the fact that the unique `.kernel` ELF section produce by the tool mixes codes and non-code, which should not impair decompiling.

[image]

Regards,

DrmnSamoLiu, 2020-08-07:
@marin-m Thank you for your swift response! It's great to know that the output file has no problem. Now it's my job to figure out how to make ghidra analyze it properly!

closed 2026-03-09 by marin-m