marin-m/vmlinux-to-elf #23: raw elf format for embedded devices

danyhm, 2020-12-11:
Hello,

I tried to run vmlinux-to-elf for a raw elf format that I obtained from a Xilinx Zynq bootloader, but it can't detect the format.

can you please take a look at it? I'm trying to analyze it with IDA but I can't get the image right.

[bootfiles.zip](https://github.com/marin-m/vmlinux-to-elf/files/5681575/bootfiles.zip)

inside there are 3 files :
boot.bin: it's a Xilinx bootloader with their own format which is generated with their "bootgen" utility.
here is the [pdf ](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_2/ug1283-bootgen-user-guide.pdf)from their website explaining the format.
based on the file I made a tool to extract the partitions from it and I got 2 .elf files.
the fsbl_nand.elf file is the first stage bootloader and the other elf file is the secondary bootloader which handles the firmware updates and safe mode and launches the main application.

here is the tool I made to analyze the file and extract the elf images: it's in java `(java -jar XilinxBIFTool.jar boot.bin)`
[XilinxBIFTool.zip](https://github.com/marin-m/vmlinux-to-elf/files/5681599/XilinxBIFTool.zip)

If I binwalk the bigger elf file I get some gzip sections but I don't know if that's right.

I also tried `arm-none-eabi-objcopy --input-target=binary --output-target=elf32-littlearm fsbl_nand.elf test.elf`
and tried to verify it with 
`arm-none-eabi-readelf -a test.elf`
it does give some information however still not a valid elf file.  i think the objcopy can't verify the input.

do you know any other way to build the final elf file?
any guidance is greatly appreciated!

skochinsky, 2020-12-11:
The files do not use ELF format (even modified) and do not seem to contain any Linux kernels but simply raw ARM (LE) code of the bootloaders. 

The u-boot file seems to be compiled for the load address 0x4000000.

danyhm, 2020-12-11:
> 
> 
> The files do not use ELF format (even modified) and do not seem to contain any Linux kernels but simply raw ARM (LE) code of the bootloaders.

that's very strange. both image names end in elf format.

> The u-boot file seems to be compiled for the load address 0x4000000.

yes, that image is copied into ram and executed from that address. I tried IDA with arm-littleendian but still can't analyze the files. 
there must be something missing.

closed 2020-12-13 by danyhm