marin-m/vmlinux-to-elf #24: Problem with Android 10/11 Xiaomi device

TheScarastic, 2021-01-01:
I am facing assertion error with xiaomi k20pro 

'Traceback (most recent call last):
  File "./kallsyms-finder", line 1116, in <module>
    kallsyms = KallsymsFinder(obtain_raw_kernel_from_file(kernel_bin.read()), args.bit_size)
  File "./kallsyms-finder", line 209, in __init__
    self.find_kallsyms_names()
  File "./kallsyms-finder", line 775, in find_kallsyms_names
    assert position > 0
AssertionError'

i printed a few values and those were

kallsyms_markers_entries = 8034917443174400
self.offset_table_element_size = 8 
position = -8034917414015472

marin-m, 2021-01-01:
Hello,

It was because there was an extra (apparently Qualcomm-specific) kernel header of 20 bytes at the beginning of the file (`UNCOMPRESSED_IMG` + a four-byte file offset to a DTB specification) that made the image contents no longer aligned on 8 bytes, even though it was present uncompressed.

I have added support for stripping this header when present in commit 9f1d6ca. The kernel is now successfully processed.

Regards.,

closed 2021-01-01 by marin-m

TheScarastic, 2021-01-11:
@marin-m sorry to trouble again but issue is back again with latest builds here is the link of boot.img https://drive.google.com/uc?id=1kHwXHJltHRm39AnVrRKu_t4n0CLTXYgu&export=download

marin-m, 2021-01-11:
Hello,

This is not a Linux kernel, this a full Android boot image. You can understand that is is the case through using a file type identifying command such as `file`, a signature searcher such as `binwalk` or an hexadecimal editor such as `hexedit`.

```shell-session
$ file /tmp/boot.img
/tmp/boot.img: Android bootimg, kernel (0x8000), page size: 4096, cmdline (console=ttyMSM0,115200n8 earlycon=msm_geni_serial,0xa90000 androidboot.hardware=qcom androidboo)
```

For instructions on how to unpack an Android boot image in order to extract the kernel, you may find this useful: https://unix.stackexchange.com/questions/64628/how-to-extract-boot-img

Regards,

TheScarastic, 2021-01-11:
@marin-m i tried on kernel as well but issue remained same

TheScarastic, 2021-01-11:
incase you want extracted kernel here it is https://git.rip/dumps/xiaomi/raphael/-/blob/raphael-user-11-RKQ1.200826.002-21.1.8-release-keys/bootimg/kernel

TheScarastic, 2021-01-11:
nvm sorry locally didn't pulled to latest commit 😢