marin-m/vmlinux-to-elf #25: The architecture of your kernel could not be guessed successfully.

orz0000, 2021-03-21:
```
$ vmlinux-to-elf vmlinuz-3.13.11-1-amd64-vyos vyos.elf
[+] Kernel successfully decompressed in-memory (the offsets that follow will be given relative to the decompressed binary)
[+] Version string: Linux version 3.13.11-1-amd64-vyos (jenkins@squeeze64devel) (gcc version 4.4.5 (Debian 4.4.5-8) ) #1 SMP Wed Aug 12 02:08:05 UTC 2015
[!] The architecture of your kernel could not be guessed successfully. Please specify the --e-machine and --bit-size arguments manually (use --help for their precise specification).
```
I'm going to analyse kernel of vyos on IDA pro, But get the above error. I wanna know why. Is that about the wired linux version that vmlinux-to-elf cannot identify? The following is the information of "vmlinuz-3.13.11-1-amd64-vyos"
```
$ file vmlinuz-3.13.11-1-amd64-vyos
vmlinuz-3.13.11-1-amd64-vyos: Linux kernel x86 boot executable bzImage, version 3.13.11-1-amd64-vyos (jenkins@squeeze64devel) #1 SMP Wed Aug 12 02:08:05 UTC 2015, RO-rootFS, swap_dev 0x3, Normal VGA
```

skochinsky, 2021-03-21:
See https://github.com/marin-m/vmlinux-to-elf/blob/master/vmlinux_to_elf/architecture_detecter.py for currently used code patterns. Probably your kernel doesn't match the pattern so you need to specify the machine type manually.

marin-m, 2021-03-21:
Hello,

Yes, more precisely, it is the following pattern that should be sought for in your decompressed kernel (you may obtain a decompressed blob with the `binwalk -e` command, it should create a file that contains the kernel version string that is mentioned in the output that you pasted):

```
e8 xxxxxxxx 55 48 89 E5
````

Which translates to:

```
call xxxxxxxx
push rbp
mov rbp, rsp
```

Currently, the architecture is considered to be correctly guessed when at least 100 prologues are present.

It may be useful to share your kernel for more investigation, or to tell which pattern your kernel uses for function prologues instead.

Regards;

skochinsky, 2021-03-21:
@marin-m I think the first call is only present in kernels compiled with profiling which is not always the case. I would maybe key on `55 48 89 E5` combined with `5D C3` (pop rbp; retn). Another option is to check for the PE header present in UEFI-supporting kernels.