marin-m/vmlinux-to-elf #27: LZ4 legacy format not recognised

qwerty-tois, 2021-04-16:
I've seen a zImage which makes use of LZ4 compression which doesn't decompress with this tool, however using unlz4 I can decompress it and then successfully use vmlinux-to-elf on resulting uncompressed image (unlz4 does claim to produce an error, however the image it produces looks complete and works fine).

From a bit of investigation it appears that the Kernel makes use the legacy LZ4 format. The magic for this is different to the LZ4 magic specified in vmlinux-to-elf source (and I guess might not also be support by the python LZ4 library?). 

https://manpages.debian.org/testing/lz4/unlz4.1.en.html
See the -l option (it actually calls out the Linux kernel).

https://github.com/cockroachdb/c-lz4/blob/master/internal/programs/lz4io.c
This contains the magic as 0x184C2102

I can't provide the zImage but it is an ARM Linux 3.12.25 image

marin-m, 2021-04-17:
Hello,

Support added in commit 6225011. Let me know if it works well. Thanks!

qwerty-tois, 2021-04-19:
Brilliant, worked a treat. Thanks for that, and thanks for the great tool!

closed 2021-04-19 by qwerty-tois