marin-m/vmlinux-to-elf #28: Android 5.4 kernel error

tiann, 2021-07-23:
```txt
Traceback (most recent call last):
  File "./main.py", line 67, in <module>
    args.base_address, args.file_offset
  File "/Users/weishu/dev/github/vmlinux-to-elf/vmlinux_to_elf/elf_symbolizer.py", line 44, in __init__
    kallsyms_finder = KallsymsFinder(file_contents, bit_size)
  File "/Users/weishu/dev/github/vmlinux-to-elf/vmlinux_to_elf/kallsyms_finder.py", line 216, in __init__
    self.parse_symbol_table()
  File "/Users/weishu/dev/github/vmlinux-to-elf/vmlinux_to_elf/kallsyms_finder.py", line 1060, in parse_symbol_table
    symbol.symbol_type = KallsymsSymbolType(symbol_name[0].upper())
  File "/Users/weishu/.pyenv/versions/3.7.3/lib/python3.7/enum.py", line 310, in __call__
    return cls.__new__(cls, value)
  File "/Users/weishu/.pyenv/versions/3.7.3/lib/python3.7/enum.py", line 564, in __new__
    raise exc
  File "/Users/weishu/.pyenv/versions/3.7.3/lib/python3.7/enum.py", line 548, in __new__
    result = cls._missing_(value)
  File "/Users/weishu/.pyenv/versions/3.7.3/lib/python3.7/enum.py", line 577, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: '1' is not a valid KallsymsSymbolType
```

It seems that the symbal_name[0] is '1' in https://github.com/marin-m/vmlinux-to-elf/blob/master/vmlinux_to_elf/kallsyms_finder.py#L1060

I try to change the code to this:

```py
            else:
                try:
                    symbol.symbol_type = KallsymsSymbolType(symbol_name[0].upper())
                    symbol.is_global = symbol_name[0].isupper()
                except:
                    logging.warn('Unknow symbol type: %s' % symbol_name[0])
                    continue
```

It can generate the elf file, but the elf seems not correct :(

Any suggestion would be helpful, thank you!

marin-m, 2021-07-23:
Hello,

Thank your for sharing. I have made a change that removes the crash but the generated image still does not seem correct. I think that it is not indented to be: the file that you shared is not a raw kernel, but an EFI image in the PE format. It should first require some ground research to be understood and possibly to integrate correct support to `vmlinux-to-elf`.

Regards,

tiann, 2021-07-23:
Thanks for your work!

Actually the kernel is extracted with [magiskboot](https://github.com/topjohnwu/Magisk) from an [Android boot image](https://drive.google.com/file/d/1LQXllyIGY2Lk2uh9Ch6-QU6-vbt_B-dn/view?usp=sharing) (xiaomi 11 with gki-5.4 kernel), maybe there are bugs in magiskboot? I will try to look into it.

tiann, 2021-07-23:
It may not be the bug of magiskboot. because i build the kernel from source, the compiled kernel output(out/arch/arm64/boot/Image) has the same issue: IDA Pro says the entry point is invalid and the function table has no symbol.

tiann, 2021-07-24:
Fixed in https://github.com/marin-m/vmlinux-to-elf/commit/acd10075a45a4ae82cb57b259b948a8336f0f524.

Thanks for your great work!

closed 2021-07-24 by tiann