marin-m/vmlinux-to-elf #29: ValueError: 2 candidates for kallsyms_token_table in kernel image

yEsPaP, 2021-08-05:
Hello, im trying to use vmlinux-to-elf on this image: https://drive.google.com/file/d/1D4tMA-gllrzNfHw6bw2DvDnIINpz6gEG/view?usp=sharing

but i get error:

```Traceback (most recent call last):
  File "./vmlinux-to-elf", line 63, in <module>
    ElfSymbolizer(
  File "/home/henris/vmlinux-to-elf/vmlinux_to_elf/elf_symbolizer.py", line 44, in __init__
    kallsyms_finder = KallsymsFinder(file_contents, bit_size)
  File "/home/henris/vmlinux-to-elf/vmlinux_to_elf/kallsyms_finder.py", line 193, in __init__
    self.find_kallsyms_token_table()
  File "/home/henris/vmlinux-to-elf/vmlinux_to_elf/kallsyms_finder.py", line 423, in find_kallsyms_token_table
    raise ValueError('%d candidates for kallsyms_token_table in kernel image' % len(candidates_offsets))
ValueError: 2 candidates for kallsyms_token_table in kernel image```

Could you help?

marin-m, 2021-08-05:
Hello,

It should now be fixed by commit 9b5d680, can you please confirm?

Thanks!

yEsPaP, 2021-08-06:
> Hello,
> 
> It should now be fixed by commit [9b5d680](https://github.com/marin-m/vmlinux-to-elf/commit/9b5d680c46c0e0a9f6e990b1b60654c772ba0256), can you please confirm?
> 
> Thanks!

Yes it does work now. Thank you

closed 2021-08-06 by yEsPaP