marin-m/vmlinux-to-elf #33: The offset of functions by kallsyms_finder.py is error

5n0wfish, 2021-12-21:
Using the kallsyms_finder.py with boot.img in this mi11 [rom](https://xiaomirom.com/download/mi-11-venus-weekly-21.5.26/#china-recovery)

```
$ python3 kallsyms_finder.py ../../../xiaomi/mi11/boot.img
[+] Kernel successfully decompressed in-memory (the offsets that follow will be given relative to the decompressed binary)
[+] Version string: Linux version 5.4.61-qgki-g0816492c5df1 (builder@c4-xm-ota-bd081.bj) (Android (6443078 based on r383902) clang version 11.0.1 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.1 (/buildbot/tmp/tmp6_m7QH b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Wed May 26 03:01:02 CST 2021
[+] Guessed architecture: aarch64 successfully in 0.00 seconds
[+] Found relocations table at file offset 0x2853a68 (count=119634)
[+] Found kernel text candidate: 0xffffffc010000000
WARNING! bad rela offset ffffffc012dd6850
[+] Found kallsyms_token_table at file offset 0x01fadb68
[+] Found kallsyms_token_index at file offset 0x01fade90
[+] Found kallsyms_markers at file offset 0x01fad0a0
[+] Found kallsyms_names at file offset 0x01bfcc18
[+] Found kallsyms_num_syms at file offset 0x01bfcc10
[i] Negative offsets overall: 0 %
[i] Null addresses overall: 0 %
[+] Found kallsyms_offsets at file offset 0x01b50a00
```

The error offset is xxx different than the correct one

5n0wfish, 2021-12-21:
`030e3000030e3003 t _head`

Ylarod, 2022-03-13:
Same issue for Mi 11 12.5.16
the offset is different from /proc/kallsyms

/proc/kallsyms and exported kallsyms: 
https://gist.github.com/Ylarod/f208c2c0949879f7acd7572f8dce37a9

kernel  download link:
https://1drv.ms/u/s!AgCZr5hFXQY7gZxSqTx1F2wKvLrD3g?e=OLzSdW

5n0wfish, 2022-03-30:
You could get over this problem by setting base address to `0xffffffc010000000`.

closed 2022-03-30 by 5n0wfish