marin-m/vmlinux-to-elf #57: ValueError: Could not guess the architecture register size for kernel

Kyle-Kyle, 2023-09-02:
This is what I get for a container-optimized-operating-system kernel (Google Cloud).
The bzImage can be downloaded from here: https://storage.googleapis.com/kernelctf-build/releases/lts-6.1.47/bzImage

Kyle-Kyle, 2023-09-02:
The full trace looks like this:
~~~
$ vmlinux-to-elf vmlinux vmlinux-sym
[+] Version string: Linux version 6.1.47 (poprdi@rbx.c.googlers.com) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40.50.20230611) # SMP PREEMPT_DYNAMIC 
[+] Guessed architecture: x86_64 successfully in 88.20 seconds
[+] Found kallsyms_token_table at file offset 0x01ed64e0
[+] Found kallsyms_token_index at file offset 0x01ed6858
Traceback (most recent call last):
  File "/home/kylebot/src/iScript/PWN_helper/linux/kernel/tools/vmlinux-to-elf", line 63, in <module>
    ElfSymbolizer(
  File "/home/kylebot/src/iScript/PWN_helper/linux/kernel/tools/vmlinux-to-elf-src/vmlinux_to_elf/elf_symbolizer.py", line 44, in __init__
    kallsyms_finder = KallsymsFinder(file_contents, bit_size)
  File "/home/kylebot/src/iScript/PWN_helper/linux/kernel/tools/vmlinux-to-elf-src/vmlinux_to_elf/kallsyms_finder.py", line 208, in __init__
    self.find_kallsyms_markers()
  File "/home/kylebot/src/iScript/PWN_helper/linux/kernel/tools/vmlinux-to-elf-src/vmlinux_to_elf/kallsyms_finder.py", line 719, in find_kallsyms_markers
    raise ValueError('Could not guess the architecture register ' +
ValueError: Could not guess the architecture register size for kernel
~~~

worstperson, 2023-09-16:
I'm not able to reproduce
```
$ ./vmlinux-to-elf bzImage bzImage.elf
[+] Kernel successfully decompressed in-memory (the offsets that follow will be given relative to the decompressed binary)
[+] Version string: Linux version 6.1.47 (poprdi@rbx.c.googlers.com) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40.50.20230611) # SMP PREEMPT_DYNAMIC 
[+] Guessed architecture: x86_64 successfully in 8.95 seconds
[+] Found kallsyms_token_table at file offset 0x01ed64e0
[+] Found kallsyms_token_index at file offset 0x01ed6858
[+] Found kallsyms_markers at file offset 0x01e8af50
[+] Found kallsyms_names at file offset 0x01d9a048
[+] Found kallsyms_num_syms at file offset 0x01d9a040
[i] Negative offsets overall: 100 %
[i] Null addresses overall: 0 %
[+] Found kallsyms_offsets at file offset 0x01d4ef60
[+] Successfully wrote the new ELF kernel to bzImage.elf
```

Kyle-Kyle, 2023-10-22:
Sorry for the late response. It still reproduces for me, even with the same command:
~~~
$ git log -1
commit fa5c9305ae1c4bbcd2debabb810e7613def690a7 (HEAD -> master, origin/master, origin/HEAD)
Author: Marin <marin-m@users.noreply.github.com>
Date:   Wed Aug 16 13:56:19 2023 +0200

    Update the supported Linux kernel version in the README file
$ md5sum bzImage 
40644ffb3a1e19690b806cd1945bc85f  bzImage
$ vmlinux-to-elf ./bzImage bzImage.elf
[+] Kernel successfully decompressed in-memory (the offsets that follow will be given relative to the decompressed binary)
[+] Version string: Linux version 6.1.47 (poprdi@rbx.c.googlers.com) (gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40.50.20230611) # SMP PREEMPT_DYNAMIC 
[+] Guessed architecture: x86_64 successfully in 5.18 seconds
[+] Found kallsyms_token_table at file offset 0x01ed64e0
[+] Found kallsyms_token_index at file offset 0x01ed6858
Traceback (most recent call last):
  File "<redacted>/vmlinux-to-elf", line 63, in <module>
    ElfSymbolizer(
  File "<redacted>/vmlinux-to-elf-src/vmlinux_to_elf/elf_symbolizer.py", line 44, in __init__
    kallsyms_finder = KallsymsFinder(file_contents, bit_size)
  File "<redacted>/vmlinux-to-elf-src/vmlinux_to_elf/kallsyms_finder.py", line 208, in __init__
    self.find_kallsyms_markers()
  File "<redacted>/vmlinux-to-elf-src/vmlinux_to_elf/kallsyms_finder.py", line 719, in find_kallsyms_markers
    raise ValueError('Could not guess the architecture register ' +
ValueError: Could not guess the architecture register size for kernel
~~~
In this case, I didn't unpack it in the first place, so I should be fine?

Kyle-Kyle, 2023-10-22:
it turns out I wasn't using the latest version. Sorry for the confusion.

closed 2023-10-22 by Kyle-Kyle