marin-m/vmlinux-to-elf #60: not working for android xiaomi kernel.

R3n3r0, 2024-01-15:


BZHugs, 2025-03-04:
Can you provide more information ?

Which kernel ? which xiaomi model ?

RickyCong, 2025-03-06:
> Can you provide more information ?
> 
> Which kernel ? which xiaomi model ?

Hello it comes from Xiaomi k70pro device which uses rom core is
6.1.75-android14-11-g16c5f6cd5e9b-ab12268515

The rom version can be downloaded here
https://xiaomirom.com/rom/redmi-k70-pro-manet-china-fastboot-recovery-rom/

Please download this refresh package
manet_images_OS2.0.101.0.VNMCNXM_20250108.0000.00_15.0_cn_8354004e94.tgz

Here I first upload two kernels and use vmlinux-to-elf to extract two binaries in different states before and after extraction
The repaired vmlinux-to-elf binary ends with the.elf suffix
And they are

### Original factory kernel binary
./vmlinux-to-elf raw_kernel raw_kernel.elf
[+] Version string: Linux version 6.1.75-android14-11-g16c5f6cd5e9b-ab12268515 (build-user@build-host) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #1 SMP PREEMPT Fri Aug 23 03:08:10 UTC 2024
[+] Guessed architecture: aarch64 successfully in 0.00 seconds
[+] Found kallsyms_token_table at file offset 0x0145c2d0
[+] Found kallsyms_token_index at file offset 0x0145c658
[+] Found kallsyms_markers at file offset 0x01412f58
[+] Found kallsyms_names at file offset 0x012c7370
[+] Found kallsyms_num_syms at file offset 0x012c7368
[i] Null addresses overall: 19.9801 %
[+] Found kallsyms_addresses at file offset 0x01204fc8
[+] Base address fallback, using first_symbol_virtual_address (ffffffc00a0b1000)
[+] Successfully wrote the new ELF kernel to raw_kernel.elf

### Kernel modified with APatch tool
./vmlinux-to-elf k70pro_hyperos_2_apatch_kernel k70pro_hyperos_2_apatch_kernel.elf
[+] Version string: Linux version 6.1.75-android14-11-g16c5f6cd5e9b-ab12268515 (build-user@build-host) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #1 SMP PREEMPT Fri Aug 23 03:08:10 UTC 2024
[+] Guessed architecture: aarch64 successfully in 0.00 seconds
[+] Found kallsyms_token_table at file offset 0x0145c2d0
[+] Found kallsyms_token_index at file offset 0x0145c658
[+] Found kallsyms_markers at file offset 0x01412f58
[+] Found kallsyms_names at file offset 0x012c7370
[+] Found kallsyms_num_syms at file offset 0x012c7368
[i] Null addresses overall: 19.9801 %
[+] Found kallsyms_addresses at file offset 0x01204fc8
[+] Base address fallback, using first_symbol_virtual_address (ffffffc00a0b1000)
[+] Successfully wrote the new ELF kernel to k70pro_hyperos_2_apatch_kernel.elf

Since github can't seem to upload this large file, I put the file in Google Cloud Drive
https://drive.google.com/drive/folders/1uxeaZUaL7-pF57C1JXpaUyKj-bxu_Jup?usp=sharing

A current guess is that Xiaomi has modified a large number of source code of the kernel, resulting in incomplete repair of this kernel, and it feels like he lost a lot of symbol information

And then just clicking on any function like proc_pid_status is going to be the status of IMPORT

[image]
[image]