marin-m/vmlinux-to-elf #61: Could not find kallsyms_names for Ubuntu Kernel

Kyle-Kyle, 2024-02-27:
I receive the following error when trying to use the project on Linux v6.5.0, which is Ubuntu's kernel.
```$ ./vmlinux-to-elf ./vmlinuz-6.5.0-17-generic vmlinux
[+] Kernel successfully decompressed in-memory (the offsets that follow will be given relative to the decompressed binary)
[+] Version string: Linux version 6.5.0-17-generic (buildd@lcy02-amd64-034) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.41) #17-Ubuntu SMP PREEMPT_DYNAMIC  (Ubuntu 6.5.0-17.17-generic 6.5.8)
[+] Guessed architecture: x86_64 successfully in 5.91 seconds
[+] Found kallsyms_token_table at file offset 0x01a27688
[+] Found kallsyms_token_index at file offset 0x01a27a00
[+] Found kallsyms_markers at file offset 0x01a26ab0
Traceback (most recent call last):
  File "/vmlinux-to-elf/./vmlinux-to-elf", line 63, in <module>
    ElfSymbolizer(
  File "/vmlinux-to-elf/vmlinux_to_elf/elf_symbolizer.py", line 44, in __init__
    kallsyms_finder = KallsymsFinder(file_contents, bit_size)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/vmlinux-to-elf/vmlinux_to_elf/kallsyms_finder.py", line 212, in __init__
    self.find_kallsyms_num_syms()
  File "/vmlinux-to-elf/vmlinux_to_elf/kallsyms_finder.py", line 774, in find_kallsyms_num_syms
    raise ValueError('Could not find kallsyms_names')
ValueError: Could not find kallsyms_names
```

Kyle-Kyle, 2024-02-27:
the file is attached here.
[vmlinuz.tar.gz](https://github.com/marin-m/vmlinux-to-elf/files/14413527/vmlinuz.tar.gz)

0ddc0de, 2024-06-03:
Yo @Kyle-Kyle,

I ran into the same issue for an 4.4.23 kernel extracted from an Android phone.
Did you manage to work around this?

Kyle-Kyle, 2024-06-05:
@0ddc0de yoyo!
Unfortunately, I don't remember what happened afterwards. My guess is that I didn't solve the issue..

PR #77 (2025-08-22): Add support for big symbols
kallsyms_names has support for symbols longer than 128 chars using ULEB128 encoding ([here](https://github.com/torvalds/linux/blob/068a56e5/scripts/kallsyms.c#L392)). 
This assumes that kernels before this change didn't have symbols of length 128-255, which is unlikely since it's only really Rust that gets these long symbols.

Fixes #61
closed 2025-08-22 by marin-m